chipsalliance/f4pga-v2x #22: Changing outputs of the `vlog_to_pbtype` function in Python 3.5

glatosinski, 2019-11-20:
I introduced tests in https://github.com/SymbiFlow/python-symbiflow-v2x/pull/18 that are running correctly on Python 3.7 and Python 3.6. However, those tests are failing in Python 3.5.

In Python 3.5, some tests are failing "randomly" - the number of failed tests differ between consecutive runs. The errors are usually some caused by moved lines  within the files, or swapped variable order, i.e.:
```
E           <?xml version="1.0"?>                                                                                                                                                                                                                                                         
E           <pb_type xmlns:xi="http://www.w3.org/2001/XInclude" name="PAIR" num_pb="1">                                                                                                                                                                                                   
E             <clock name="CLK" num_pins="1"/>                                                                                                                                                                                                                                            
E             <input name="I" num_pins="4"/>                                                                                                                                                                                                                                              
E             <output name="O" num_pins="1"/>                                                                                                                                                                                                                                             
E             <pb_type blif_model=".subckt DFF" name="ff" num_pb="1">                                                                                                                                                                                                                     
E               <clock name="CLK" num_pins="1"/>                                                                                                                                                                                                                                          
E               <input name="D" num_pins="1"/>                                                                                                                                                                                                                                            
E               <output name="Q" num_pins="1"/>                                                                                                                                                                                                                                           
E               <T_setup clock="CLK" port="ff.D" value="10e-12"/>                                                                                                                                                                                                                         
E               <T_hold clock="CLK" port="ff.D" value="10e-12"/>                                                                                                                                                                                                                          
E               <T_clock_to_Q clock="CLK" max="10e-12" port="ff.Q"/>                                                                                                                                                                                                                      
E             </pb_type>                                                                                                                                                                                                                                                                  
E             <pb_type blif_model=".subckt LUT4" name="lut" num_pb="1">                                                                                                                                                                                                                   
E               <input name="I" num_pins="4"/>                                                                                                                                                                                                                                            
E               <output name="O" num_pins="1"/>                                                                                                                                                                                                                                           
E               <delay_matrix in_port="lut.I" out_port="lut.O" type="max">           
E           30e-12 20e-12 11e-12 3e-12                                                                                                       
E           </delay_matrix>                                                                                                                  
E             </pb_type>                                                                                                                     
E             <interconnect>                                                                                                                 
E         -     <mux input="ff.Q lut.O" name="mux" output="PAIR.O">   
E         ?                 -----                                     
E         +     <mux input="lut.O ff.Q" name="mux" output="PAIR.O">
E         ?                      +++++                                                                                                       
E                 <metadata>                                          
E                   <meta name="fasm_mux">                                                                                                   
E         + lut.O : L                                                 
E           ff.Q : F                                                                                                                                                                                                                                                                      
E         - lut.O : L                                                 
E           </meta>                                                   
E                   <meta name="type">bel</meta>                                                                                             
E                   <meta name="subtype">routing</meta>
E                 </metadata>                                         
E               </mux>                                                
E               <direct input="PAIR.CLK" name="ff-CLK" output="ff.CLK"/>                                                                     
E               <direct input="lut.O" name="ff-D" output="ff.D">                                                                             
E                 <pack_pattern in_port="lut.O" name="pack-LUT2FF" out_port="ff.D"/>
E               </direct>                                             
E               <direct input="PAIR.I[0]" name="lut-I[0]" output="lut.I[0]"/>                                                                
E               <direct input="PAIR.I[1]" name="lut-I[1]" output="lut.I[1]"/>                                                                
E               <direct input="PAIR.I[2]" name="lut-I[2]" output="lut.I[2]"/>                                                                
E               <direct input="PAIR.I[3]" name="lut-I[3]" output="lut.I[3]"/>                                                                
E             </interconnect>                                         
E           </pb_type>                                                                                                                       
```
```
E           <?xml version="1.0"?>                                                                                                                                                                                                                                                         
[... cut ...]
This issue does not occur in Python 3.6 and Python 3.7. I found a discussion in https://bugs.python.org/issue34160 suggesting that this may be result of quite significant changes in the Python dictionaries between 3.5 and 3.6.

Links to examples of failing pipelines:
https://travis-ci.com/SymbiFlow/python-symbiflow-v2x/jobs/258554284
https://travis-ci.com/SymbiFlow/python-symbiflow-v2x/jobs/258523115
https://travis-ci.com/SymbiFlow/python-symbiflow-v2x/jobs/258519609
https://travis-ci.com/SymbiFlow/python-symbiflow-v2x/jobs/258288841

mithro, 2019-11-20:
@glatosinski Sounds like you are not sorting the output? The following XSL in architecture definitions does the sorting -> https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/common/xml/sort-tags.xsl

glatosinski, 2019-11-21:
I apply the conversion from `vtr_xml_utils`, which also includes the `sort-tags.xsl` stylesheet. I looked at the sort-tags.xsl and discovered that there is no rule for `delay_constant`, so in fact this part of the output was not sorted. I created a pull request in `vtr_xml_utils tool` that adds the rule for `delay_constant` in https://github.com/SymbiFlow/vtr-xml-utils/pull/6.

As for `mux`, I see that there is already a rule for sorting this by `output`, `name` and `input`, but the problem here is in the value of the `input` attribute, which has unsorted space-separated values.

mithro, 2019-11-22:
@glatosinski I think you should be able to sort the space-separated values with xslt? Maybe something at https://stackoverflow.com/questions/41872791/xsl-for-each-on-space-separated-attribute-values helps?