chipsalliance/f4pga-v2x #40: Missing golden.pb_type.xml in some test cases

daniellimws, 2020-04-14:
A lot of test cases in [tests/clocks](https://github.com/SymbiFlow/python-symbiflow-v2x/tree/master/tests/clocks) and the one in [tests/simple_pll](https://github.com/SymbiFlow/python-symbiflow-v2x/tree/master/tests/simple_pll) do not have golden.pb_type.xml. 

Are these test cases incomplete?

daniellimws, 2020-04-14:
Additionally, the names for modules in tests/clocks are `module BLOCK` instead of `module FILENAME`. This causes pytest to fail when a golden.pb_type.xml file is added.

mithro, 2020-04-15:
@mkurc-ant - Do you know if these where a WIP or?

mithro, 2020-04-15:
Looks like they might have come in via https://github.com/SymbiFlow/symbiflow-arch-defs/pull/316/files which was written by @acomodi?

mkurc-ant, 2020-04-15:
@mithro I can fix these tests. Some tests do not have either golden model or pb_type as they are meant to test only the one that is present.