chipsalliance/f4pga-v2x #47: Figure out how to skip copyright header in verilog output in rendered Sphinx docs

mithro, 2020-04-16:
It would be nice to automatically skip the copyright header when displaying the verilog output into the Sphinx documentation.

Currently a verilog file like this;
``verilog
/*
 * Copyright (C) 2020  The SymbiFlow Authors.
 *
 * Use of this source code is governed by a ISC-style
 * license that can be found in the LICENSE file or at
 * https://opensource.org/licenses/ISC
 *
 * SPDX-License-Identifier:	ISC
 */

/*
 * `input wire a` should be detected as a clock because of the `(* CLOCK *)`
 * attribute.
 */
(* whitebox *)
module BLOCK(a, b, o);
	(* CLOCK *)
	input wire a;
	input wire b;
	output wire o;
endmodule
```

You end up with the output which looks like this;
![Screenshot from 2020-04-16 10-42-23](https://user-images.githubusercontent.com/21212/79488694-fd6d2300-7fce-11ea-896b-2738347e51eb.png)

This is probably a feature for sphinxcontrib-verilog-diagrams?

daniellimws, 2020-04-17:
I managed to hide it with the `:start-after:` option of `literalinclude` as seen in https://github.com/SymbiFlow/python-symbiflow-v2x/pull/42/commits/646d3d249a95f8aae65efe8bfa51e2f7ee4f203e.

mithro, 2020-04-23:
@daniellimws - It would be nice if there was a `literalinclude` like directive that automatically detected the copyright header in the file and added the `:start-after:` bit automatically. This could probably be done via a similar manner to https://github.com/mithro/sphinxcontrib-session

daniellimws, 2020-04-23:
@mithro Where would be a good place to implement this directive? Shall I add this feature into https://github.com/mithro/sphinxcontrib-session?

mithro, 2020-04-23:
@daniellimws I think as part of `sphinxcontrib-verilog-diagrams` probably makes some sense? Otherwise maybe we need a new `sphinxcontrib-fancyinclude` type project?

daniellimws, 2020-04-23:
Alright sounds good, I'll add it to `sphinxcontrib-verilog-diagrams`.

closed 2020-05-21 by mithro