chipsalliance/f4pga-v2x #72: CI failures for Python <3.6

mkurc-ant, 2020-09-10:
There are CI failures for Python version <3.6. I've verified them with the https://github.com/SymbiFlow/python-symbiflow-v2x/pull/32 PR:

Python 2.7 (I guess we are actually not interested in Python 2):
```
py27 run-test: commands[2] | flake8 .
./v2x/vlog_to_pbtype.py:154:20: E999 SyntaxError: invalid syntax
./v2x/mux_gen.py:173:25: E999 SyntaxError: invalid syntax
./v2x/yosys/json.py:281:35: E999 SyntaxError: invalid syntax
```

Python 3.4  (Incompatible lxml):
```
    ERROR: This lxml version requires Python 2.7, 3.5 or later.
```

Python 3.5 - Here are a few test failures for V2X itself. I guess that it has to do that Python 3.6 introduces some data order preservation mechanisms not present in 3.5. Fixing that requires more thorough investigation and I don't think that it is the highest priority as we mostly use Python 3.6/3.7.

mkurc-ant, 2020-09-11:
Failures for 3.5 have been fixed in https://github.com/SymbiFlow/python-symbiflow-v2x/pull/76

mithro, 2020-09-11:
Let's definitely drop 2.7 and 3.4

We can probably drop 3.5 too

mkurc-ant, 2020-09-14:
With https://github.com/SymbiFlow/python-symbiflow-v2x/pull/76 2.7 and 3.4 are disabled in CI. 3.5 is still enabled as I've managed to fix issues with maintaining item order. So shall I disable that one too?

mkurc-ant, 2020-09-17:
CI passes for Python 3.5, 3.6, and 3.7. Other versions are dropped. Closing.

closed 2020-09-17 by mkurc-ant