chipsalliance/f4pga-v2x #91: FF with clock enable and syncronous set/reset does not work

litghost, 2020-11-30:
```
(* whitebox *)                                                                                                                                                                                                                                                                            
module FF(C, D, Q, CE, SR);                                                    
    parameter RESET_VALUE = 1'b0;                                                
                                                                               
    (* CLOCK *)                                                                
    input C;                                                                   
                                                                               
    (* SETUP = "C 10e-12" *)                                                   
    input D;                                                                   
                                                                               
    (* SETUP = "C 10e-12" *)                                                   
    input CE;                                                                  
                                                                               
    (* SETUP = "C 10e-12" *)                                                   
    input SR;                                                                  
                                                                               
    (* CLK_TO_Q = "C 10e-12" *)                                                
    output reg Q;                                                              
                                                                               
    always @(posedge C) begin                                                  
        if(SR) begin                                                           
            Q <= RESET_VALUE;                                                  
        end else if(CE) begin                                                  
            Q <= D;                                                            
        end                                                                    
    end                                                                        
                                                                               
endmodule    
```

This generates no clock annotations in the `.model.xml` for D/CE/SR.  None of the current examples in the v2x documentation show how to make this work properly.

mithro, 2020-11-30:
FYI - @mkurc-ant

mithro, 2020-11-30:
https://python-symbiflow-v2x.readthedocs.io/en/latest/examples/vtr/dff/README.html#timings

mithro, 2020-11-30:
BTW - Your example looks valid and should probably also be a test case.