chipsalliance/f4pga-v2x #114: Why v2x don't support routing mux which has output bus?

KKtiandao, 2022-04-25:
I view the code of function make_container_pb. The 'sinks' of routing mux is an array struct.  But there are some 'assert' to judge if routing mux has multi ouputs. So in the case of mux with bus in and output, the 'assert ' will be false. 
[image]
https://github.com/chipsalliance/f4pga-v2x/blob/11bc48d06352250bb878d42eedefdc6bbaa6e402/v2x/vlog_to_pbtype.py#L672

kgugala, 2022-04-25:
According to the VPR docs (https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cmuxname=) buses are not supported. There is a note saying:
```
Buses are not yet supported so all muxes must use one bit wide data only!
```

The assert in v2x is to guard exactly this. If this is not true anymore we should update VPR docs (and v2x to support this)

KKtiandao, 2022-04-26:
@kgugala Thanks. I have another questions. Can v2x  generate the pb_type subtag on timing like <T_setup> <T_hold> ?

mithro, 2022-04-26:
See https://python-symbiflow-v2x.readthedocs.io/en/latest/examples/vtr/dff/README.html

KKtiandao, 2022-04-27:
@mithro Thanks!

closed 2022-04-27 by KKtiandao