chipsalliance/f4pga-v2x #117: ReadTheDocs page not linking to GitHub repository

mithro, 2022-06-03:
The docs at https://python-symbiflow-v2x.readthedocs.io/en/latest/ don't link back to the repository.

mithro, 2022-06-03:
@umarcor - Probably an incorrect configuration of ReadTheDocs?

umarcor, 2022-06-06:
@mithro I don't have access to RTD projects. However, the documentation of this repo seems to be https://f4pga-v2x.readthedocs.io/, not https://python-symbiflow-v2x.readthedocs.io.

/cc @kgugala

mithro, 2022-06-06:
@umarcor - We should push a redirect to the https://python-symbiflow-v2x.readthedocs.io/ location?

umarcor, 2022-06-06:
@mithro I believe that a redirect requires explicit communication with RTD maintainers. In other repos we did add an admonition to the "prolog". See https://symbiflow.readthedocs.io or https://symbiflow-examples.readthedocs.io.